Semi-ATE/Project-Template #14: Q64.10x0E

nerohmot, 2021-06-08:
Please remove the exposed pad to the bottom (it is on the top of the package)

nerohmot, 2021-06-08:
By the way ... the Qxxxxx packages I remember are for Saturn and Jupiter, we exchanged the Jupiter for the AD5560 ... and still the package is Qxxx ??? shouldn't that be SV-64-3

nerohmot, 2021-06-09:
[Jupiter.pdf](https://github.com/Semi-ATE/Project-Template/files/6621918/Jupiter.pdf) at page 7

nerohmot, 2021-06-09:
But we actually need the footprint of the AD5560 : https://www.analog.com/media/en/technical-documentation/data-sheets/AD5560.pdf page 65/TOP ➜ name should be SV-64-3 and in the schematic library of the AD5560 the footprint should become SV-64-3

nerohmot, 2021-06-09:
Hi Daniel, please find here the correct 3D model for the SV-64-3

[SV-64-3.STEP.zip](https://github.com/Semi-ATE/Project-Template/files/6622989/SV-64-3.STEP.zip)

Daneca, 2021-06-09:
Hi Tom,

For the AD5560 I'm done, but the second component is actually Saturn and not Jupiter. Please share me a pdf of the Saturn.
Thanks!

Daneca, 2021-06-09:
As confirmed over a slack call, the Saturn footprint is OK and doesn't need any attention.
The AD5560 is also done.

PR #26 (2021-06-09): Daniel june 08 03
Issues Closes #10, Closes #14, Closes #15, Closes #16, Closes #17, Closes #19, Closes #20, Closes #21, Closes #22, Closes #23.
Updated SM and PM for footprints CP-72-4, CP-64-3,LFCSP32,CP-16-17,CP-10-9,SC70-6,MSOP8(RM-8),CP-8-4,CP-16-17,T2044-3,Q128.14X20B,Q64.10X10E,PVSN10 and MDT420M01001.
Updated MiniSCT connector, spacer and created guidance and nut.
Updated schematic library.
closed 2021-06-09 by Daneca